TcOpenGroup/tcopen-app-templates #102: [BUG] Clean cycle times show sequence cycle times

KrisztianOlah, 2024-01-18:
**Describe the bug**
Clean cycle time should show the elapsed time from calling StartCleanCycleTime() instead it shows SequenceElapesedTime.

CuBase.UpdateInfo() :
[image]

That GroundTask should be AutomatTask, this will fix what is displayed. Though a better solution would be:

[image]

Furthermore, _startCleanCycleTime and _stopCleanCycleTime should be reset when ground is called and when the Automat sequence finishes or when it starts each iteration. OnStart did not work for me so please advise. Thanks

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Microsoft Windows [Version 10.0.18363.1082]]
 - Visual Studio version [e.g. Microsoft Visual Studio Professional 2019 Version 16.7.3]
 - TwinCAT3 version [eg. 4024.10]
 - TcOpen Version [e.g. 0.6.1]

**Additional context**
Add any other context about the problem here.

KrisztianOlah, 2024-01-19:
After a second look Resetting _startCleanCycleTime and _stopCleanCycleTime in TcoTaskSequencer.CompleteSequence() will reset each iteration.

KrisztianOlah, 2024-01-19:
/cib

peterbarancek, 2024-01-29:
closes #112

peterbarancek, 2024-01-29:
closes #112