TcOpenGroup/tcopen-app-templates #103: [BUG] Average cycle time shows incorrect value

KrisztianOlah, 2024-01-19:
**Describe the bug**
Average cycle time shows about tenth of the correct value. 

The problem is in TcoTaskedSequencer. StopCleanCycleTime() :

[image]

On the first cycle StopCleanCycleTime is called _LastCycleTime is T#0MS as it will only be written to by CompleteSequence which is generally called after StopCleanCycleTime. Also for the same reason there is no point calculating _cycleTimeAnalitics.Average, but we need _cycleTimeAnalitics.AverageCleanCycle.

KrisztianOlah, 2024-01-19:
/cib

peterbarancek, 2024-01-29:
closes #112

closed 2024-01-29 by peterbarancek